methlabUZH/automagic #5: Default value

AlexChalard, 2019-07-22:
Hello,

At first, I would like to thank you for this huge piece of work contributing to the eeg community.

I would like to have some feedback concerning user and your experience concerning the value of the ASR parameters, do the defaults values are a good compromise between preserving brain signals and remove artifacts ? 

Thank you

Kind regards

Alex

nickilanger, 2019-07-23:
Dear Alex, 

Thank you for your interest in Automagic. I think it's difficult to give a general answer to your question. It always depends on the artifacts in your data, the signal-to-noise ratio, the amount of good quality time segments. The best advice I can give: do a visual inspection and check if the artifacts are sufficiently removed. 

Note that the selection of these two additional processing steps has consequences for the original EEG as the EEG dataset might be shortened by the time periods that could not be adequately recovered (therefore we don't use ASR as default, but rather use after Automagic an additional amplitude threshold to identify bad segments). In a limited set of tests with small data sets, we observed that the time periods rejected by ASR varied and hence were not perfectly replicable. Therefore, we suggest testing the use of ASR with respective data sets.

All the best, 

Nicolas

AlexChalard, 2019-07-23:
Dear Nicolas,

Thank you for your reactivity.

Precisely I want to try the ASR because at times I have burst that I am forced to remove on a dataset with little trials. 
I wanted to know how the data cut by the ASR was organized, does automagic clearly indicate this and where the data was cut? 
For example, if my data has been cut in the middle, the algorithm replaces the middle with NaN, or does it cut and change the latency of the total signal? 

Alex

nickilanger, 2019-07-23:
Hi Alex, 

If, I’m correctly informed, ASR does change the latency and does not replace bad segments with NaN. But you can check that yourself, by checking the latencies (e.g. EEG.events) before and after ASR/automagic preprocessing. Please note, that we have not changed the original function at All. So you should find the exact information about ASR here:

https://github.com/sccn/clean_rawdata/blob/master/readme-asr.txt 

Best, 

Nicolas

closed 2020-02-10 by JohnSmallridge