methlabUZH/automagic #6: Prep processing time

AlexChalard, 2019-07-24:
Hello everyone,

I tried to process the data using the Prep pipeline, I already used PREP in the past and I know it's computationally intensive, but using Automagic after 20 minutes of parpool it still hasn't progressed and remains blocked, does it work well for you?  

Thank you

AlexChalard, 2019-07-24:
It seems that Matlab R2017b introduce an error when parfor is running.

nickilanger, 2019-07-24:
Hi Alex, 

For the first subject (and probably only the first time) you have to manually enable MATLAB parallel processing. 

We had some issues with parfor on some instances as well. Yes, maybe try a newer MATLAB version. 

Please note, PREP is unchanged in Automagic. So, it should have the identical computation time as when you use PREP outside of Automagic. Let us now, how if you could solve the issue with a newer version of MATLAB.

Best, 

Nicolas

AlexChalard, 2019-07-24:
Hi Nicolas,

I just moved on Matlab R2017a to r2019a and it works.
I found that some R2017a users had idletimeout issues during the parfor processing in some cases.
Anyway, newer version of matlab fix the issue for me.

Alex

nickilanger, 2019-07-24:
Awesome! Thanks for presenting a solution for the problem. This is helpful for other users. 

Best, 

Nicolas

closed 2020-02-10 by JohnSmallridge