methlabUZH/automagic #7: how to add raw data

MRBMELMH, 2019-08-03:
Dear Nicolas,
        Firstly, thank you and your team created such a great work! It‘s awesome!
        I have a problem about how to add my EEG data to the menu?My data was recorded from the g.recorder, which can't be read in Automagic .And then I changed the file extension .set , it sill can't be read. Could you tell me the solution of this problem?
       Thanks  sincerely
       M.LMH

nickilanger, 2019-08-05:
Dear Ma Rui, 

Can you import the .hdf5 files into EEGLab (without using Automagic)? If so, you can simply import the files into EEGLab first and then save it as a .mat file. It seems that you saved the files as .set. The .set format is not compatible with Automagic. 

If you have imported the files with EEGLab, MATLAB creates an structures named "EEG". Save this "EEG" structure with the "save" function:

save('yourfilename.mat','EEG')

Then Automagic can definitely load the .mat file. 

Best, 

Nicolas

MRBMELMH, 2019-08-06:
Dear Nicolas,

Thank you for your reply ! I tried your suggestion,but it seems can't work.First, the 2019 version of EEGLAB can't save the .hdf5 files as .mat files , and i don't understand that you said "mported the files with EEGLab, MATLAB creates an structures named "EEG". Save this "EEG" structure with the "save" function“  how to realize, could you put a video or screeshoot about the solution?

Thanks  sincerely,

Ma Rui

MRBMELMH, 2019-08-06:
Dear  Nicolas,
     
Thanks for your reply. This problem has already be solved. 
I have another question : Can I use a 64-channel data in Automagic？
 
Thanks 
Ma Rui

nickilanger, 2019-08-06:
Dear Ma Rui, 

Nice to hear that the data import did work now. Sure, you can use 64-channel data or any other electrodes EEG system. All you need is the electrodes coordinates file for your EEG system.

MRBMELMH, 2019-08-06:
Dear Nicolas,

Thank you for your reply.
I still want to ask one more question: how can I save the result as .set file?

Thanks sincerely,
Ma Rui

nickilanger, 2019-08-06:
The gp_m2 is the final (cleaned) data set. If you load this one to MATLAB you will see a EEG and automagic structure. The EEG structure is the same as from EEGLab, so you can simple continue to work with EEGLab function (e.g. save as a .set file). 

Best, 

Nicolas

MRBMELMH, 2019-08-06:
But when I save it as .set file,and load it in EEGLAB,there only have 63 channels, I don't know which channel has been excluded.  In the Automagic processing, I didn't choose to exclude any channel.If I don't know which channel has been excluded, I can't do channel location .
[
[image]
](url)

nickilanger, 2019-08-06:
I don't know what settings you have used for automagic (which reference? did you use bad channel identification?). are you sure that there were 64 channels before Automagic. It might be that the reference channel is not loaded in a first place. You can check the channels by looking at the EEG.chanlocs structure.There you see the labels and coordinates of all used electrodes.

MRBMELMH, 2019-08-06:
This is my settings:
[
[image]
](url)
[
[image]
](url)
Actually, there are 64 EEG channels and 2 EOG channels.

nickilanger, 2019-08-07:
Did you open the raw files (before preprocessed in Automagic) in EEGlab? What's the number of channels there? or check EEG.chanlocs. 

I see in your setting that you selected two electrodes as references channel (63 and 64). Are you sure you recorded this with two reference electrodes? Are you sure they are in the raw recordings?

MRBMELMH, 2019-08-08:
YES , I chose two reference channels ,which were putting at two earlobes. 
And now ,i have another problem that the processing  is at a standstill in PREP :
[
[image]
](url)
And there are my settings:
[
[image]
[
[image]
](url)
](url)

nickilanger, 2019-08-08:
could you send me a raw file please. so i can take a look at the two reference electrode problem. 

It seems that PREP is working. It just takes a lot of processing time depending on your computer. 

Best, 

Nicolas

MRBMELMH, 2019-08-08:
OK，I'll send my raw files to you in e-mail.

Thaanks
MA RUI

nickilanger, 2019-08-09:
Hi Ma rui, 

I did no receive any raw files. 

Best,

Nicolas

Nicolas Langer
Assistant Professor of Methods of Plasticity Research
University of Zurich
Andreasstrasse 15 (Office AND.4.58)
8050, Zurich, Switzerland
n.langer@psychologie.uzh.ch
phone: (+41) 44 635 34 14
http://www.psychology.uzh.ch/en/areas/nec/plafor.html

closed 2020-02-10 by JohnSmallridge