methlabUZH/automagic #8: Index out of bounds

Mode5t, 2019-08-20:
Hi methlabUZH!
         I'm Chen,parallel computing started working.but I have 32 subjects,and some of subjects didn't pass the preprocessed stage,and throw index out of bounds exception,like this.

[image]
         ERROR
[image]
        EEG structure
        All data have 32 channels,but 
        "noChannelAsCell{nochanId}=EEG.chanlocs(g.nochannel(nochanId)).labels;"
        Exception tell me EEG.chanlocs size is 30,and different subject throw same exception,also 30 index out of bounds.

I am looking forward to your reply.
Thank you
Chen

Mode5t, 2019-08-20:
[image]
[image]

Mode5t, 2019-08-21:
Hi methlabUZH!
I found that option 'has aleardy a reference chanel' will cause index out of bounds exception in interpolation stage, when this option was chose,exception happened,if choose add a new channel,it work.some screenshot as follow:
[image]
project page
[image]
configuration page
[image]
exception
[image]
work without providing reference channel

Finally,when I provide a reference channel index,and the output data structure will remove one channel data,I think this channel should be this reference channel,but when I open it found that the index channel with all 0,I think maybe some var error during the processing.Maybe automagic remain has some bug,but it' also a great work. 

I am looking forward to your reply.
Thank you
Chen

nickilanger, 2019-08-21:
Hi Chen, 

Thanks for your messages. 

First of all we need to know, whether your data actually does contain the reference electrode or not. Do all data you want to preprocess have the reference channel and the same index. If not, you need to create seperate projects. 

If all files have a reference channel in the data (you can check them by importing it into EEGlab and see if any of of the rows in EEG.data are zero), you did the right thing to select that "has already reference channel" and then add the index. As you correctly identified that the after preprocessing the index of this channel is set to zero. The reference is always zero and should have zeros already before the preprocessing (otherwise this electrode is not the reference channel and you the data actually does not contain the reference channel?). To sum up, the preprocessed data are referenced to the original reference electrode. Thus the reference channel has to be zero. You can re-reference it to average reference or any other reference later (e.g. in EEGLAB). 

Hope this makes sense. 

Best, 

Nicolas

closed 2020-02-10 by JohnSmallridge