methlabUZH/automagic #9: Channel Location Error

m-topor, 2019-08-27:
Hi there! 

I might be coming with an unusual set up but I am hoping to be able to work around it. 

I am using the Mobita mobile EEG system produced by TMSI and supplied by Biopac. Recordings are taken by Biopac's own Acqknowledge software. The EEG recording files are saved in Acqnowledge specific format but I am able to export them to an .edf format and that's what i have been doing. 
I cannot export channel locations from Acqknowledge (or at least I am not aware how to) but I have created a channel locations file using EEGLab and exported in .ced format. 

I have tried to use Automagic with the .edf files extracted from Acqknowledge and the channel locations file created in EEGLab based on those .edf files

I noticed that there was a problem with the channels, as the original map of the MOBITA electrodes included double names for some channels e.g. "T7/T3". I have tried to get around this in different ways. 
- I adapted the channel locations file to provide parameters for channel locations with double names such as "T7/T3" - the error persisted
- I changed channel names in the original Acqknowledge file so only single labels were given to all channels, e.g. "T7". I exported the file again as .edf and then created a new channel location file in EEGLab in .ced format. - the error still persists 

I have double checked and for both solutions above the created channel locations file works well when applied to files imported to EEG lab so I am not sure where the error might be coming from. 

```Matlab
*******Start preprocessing all dataset*******
Processing file Participant01_pilot1 ...(file 1 out of 1)
Insufficient number of outputs from right hand side of equal sign to satisfy assignment.

Error in readlocs (line 348)
      g.format = chanformat(indexformat).importformat;

Error in pop_chanedit (line 664)
                        [chans] = readlocs(tmpargs{:});

Error in systemDependentParse (line 137)
            [~, EEG] = evalc(['pop_chanedit(EEG,' ...

Error in preprocess (line 133)
    systemDependentParse(data, EEGSystem, ChannelReductionParams, ...

Error in Block/preprocess (line 423)
            [EEG, fig1, fig2] = preprocess(data, self.params);

Error in Project/preprocessAll (line 456)
                [EEG, automagic] = block.preprocess();

Error in mainGUI>runpreprocessbutton_Callback (line 751)
    project.preprocessAll();

Error in gui_mainfcn (line 95)
        feval(varargin{:});

Error in mainGUI (line 61)
    gui_mainfcn(gui_State, varargin{:});

Error in
matlab.graphics.internal.figfile.FigFile/read>@(hObject,eventdata)mainGUI('runpreprocessbutton_Callback',hObject,eventdata,guidata(hObject)) 
Error while evaluating UIControl Callback.
```

All help will be very appreciated!

Thank you!
Marta

JohnSmallridge, 2019-09-10:
Hi Marta,

Thanks for your question, and I think this is an odd situation. 
Have you tried exporting the channel loc file as a different file format? Does the same error appear?

John

nickilanger, 2019-09-14:
Hi Marta, 

I would guess, that there must be something wrong with setting the reference channel. Please make sure, if a reference channel is already in the data (a row with all zeros) or a non-scalp reference channel was used (check your EEG system) or if you should add a new reference (scalp reference, but not in the EEG.data). 

Best, 

Nicolas

yjmantilla, 2020-01-29:
Hi, I had the same problem while importing locations from ced files. It appears that the error arises from an inconsistency in readlocs.m. So I'm posting my findings here if anyone else runs into the same problem:

If you put '.ced' in the "channel location file type" box in the automagic main gui you will get the following error:

`Insufficient number of outputs from right hand side of equal sign to satisfy assignment.
Error in readlocs (line 348)
      g.format = chanformat(indexformat).importformat;`

One could try converting to another sensor location format and use that one, or:

**Instead of putting '.ced' in the box, try putting '.chanedit' in it (maybe 'chanedit' without the dot will work too).**

This works because internally eeglab relates '.ced' with the filetype 'chanedit' as can be seen in readlocs.m line 328:

`case 'ced', g.filetype = 'chanedit';`

For some reason there is no conversion of 'ced' to 'chanedit' in the code for the case when one puts 'ced' in the aforementioned box.

m-topor, 2020-03-04:
Thanks for the comments on this thread! I resolved the issue by loading a .locs file instead of a .ced but thanks @yjmantilla for your explanation on how to approach the .ced problem directly. Might be handy in the future.

closed 2021-02-17 by ksgfan