methlabUZH/automagic #12: [Suggestion] ICLabel

AlexChalard, 2019-09-18:
Dear Automagic team,

Once again thank you for your wonderful work.

I am writing this message just to make a suggestion regarding the classification of ICAs, there is the IClabel algorithm that appears relevant. Is it possible / planned that this option will be added to the toolbox? 

https://www.sciencedirect.com/science/article/abs/pii/S1053811919304185

Kind regards,

Alex

nickilanger, 2019-09-18:
Dear Alex, 

Thanks for your message. But IC label (version 2) is actually included in the new developmental version of Automagic. 

Best, 

Nicolas

AlexChalard, 2019-09-18:
Dear nicolas,

I am really glas to hear that this feature will be added in the next version/

Best,

Alex

nickilanger, 2019-09-20:
You can already use it by downloading the developmental version.

AlexChalard, 2019-09-20:
Where can I found it ?

nickilanger, 2019-09-20:
if you go on the main github page. und the left side you see branch: master. switch to development. 

Best, 

Nicolas

closed 2021-02-17 by ksgfan