methlabUZH/automagic #14: elicit warning/error if chanlocs cannot be read

behinger, 2019-10-28:
Hi!
I tried using your toolbox on a biosemi 64+16 extra channel file. It's not my data and I look for a quick way to (re)analyse them. Anyway, I started running into problems using iclabel, because as I found out it didn't import my external-chanloc file.

The chanloc file was only 64 channels large, and did not contain the 16 extra channels. The eeglab function called in ["systemDependentParse.m" in L137](https://github.com/methlabUZH/automagic/blob/c6e0f2756861afbb247e55f381b40e4569f451f1/preprocessing/systemDependentParse.m#L137) silently passes over this issue. Maybe a warning there would be appropriate.

Best, Bene

nickilanger, 2019-10-29:
Hi Benedikt, 

Thanks for your interest in our preprocessing pipeline and pointing out the issue.

ICLabel is still someway in development and we simply included it as it is developed by EEGLab developers (we generally don't modify functions from other developers). I assume (have to double check) this build in "silent passing" is included in the original IC Label function. So, probably better to contact EEGLab directly. 

The channel location files has to have the same dimensions as the EEG data. How about you remove the 16 extra channels first (EEGlab function: pop_select), before your input it into Automagic. 

Best, 

Nicolas

behinger, 2019-10-29:
Hi Nicolas,
thanks for your response.

the silent passing is the EEGlab "pop_chanedit" function you call in automagic (see linked function in original response). Later IClabel "passes" with an error because it rightfully though cant generate the topographies.

I "solved" the issue by appending the extra channels to my chan loc file, removal then via automagic.

nickilanger, 2019-10-29:
I see. so, we'll try to implement a sanity check before IC Label. 

I would be careful  though with the usage of non-EEG channels for IC Label. My guess is that IC Label is trained only based on EEG channels. But worth exploring. 

Cheers

behinger, 2019-10-29:
clarification-question: I was thinking/hoping that automagic removes the to-be-removed-channels before IClabel. I think this is true given the outputted EEG file. [sidenote: any plans to let user specify ICA algorithm, i.e. amica?]

nickilanger, 2019-10-29:
yes. this is true. to be removed channels are not included in the ICA. 

We can put the specification option of the ICA algorithm on our to-do list. ;)

behinger, 2020-11-10:
haha. had exactly the same error again, sorry for closing