methlabUZH/automagic #15: bug: cannot merge subject, missing field in self.params

behinger, 2019-10-29:
Hi!
(developer version) I'm not sure why this error occured. "Block.m" in Line 223 is throwing an error because 
```
K>> autParams.Settings

ans = 

  struct with fields:

    trackAllSteps: 1
      pathToSteps: 'C:\Users\behinger\cloud\PostDoc\exercises\readinessPotential\bids_r7_results\sub-201\eeg\/allSteps_sub-201_eeg'

K>> self.params.Settings

ans = 

  struct with fields:

    trackAllSteps: 1
```
Are not equal (for obvious reasons, one field is missing). Not sure if this is a bug, or if I messed up something along the way. It could also be that selecting the "save all steps on the way" option raises this error somehow.

All other `fieldnames(autParams)` are identical

nickilanger, 2019-10-29:
Hi Benedikt, 

Can you extend your description, what exactly you did? Did you try to run the preprocessing through the command line? Can you provide the code you used?

Best, 

Nicolas

behinger, 2019-10-29:
Hi!
I actually used the gui :-) 
[image]
[image]

Pretty straight forward analysis pipeline
Best, Bene

nickilanger, 2019-10-29:
This looks straight forward indeed. :) If you send me one of your files and the channel locations file, I can test it for you. 

Best, 

Nicolas

behinger, 2019-11-01:
(sorry I missed the notification).
I analysed this dataset:
https://osf.io/s3fdv/ subject 201
with this chanloc file: 
[BioSemi64.loc.txt](https://github.com/methlabUZH/automagic/files/3797879/BioSemi64.loc.txt)

nickilanger, 2019-11-01:
Hi Benedikt, 

Thanks for the files. I'm running the preprocessing on my computer now. I will let you know, how it goes. I've noticed two things:

1. The .bdf file itself has the channel location already included (if you import it with EEGlab). But I added the channel location file anyway in Automagic. 

2. Based on your settings (print screens) you excluded channel index 80, but there are only 79 channels. You have to provide the channel index not the channel name. So, this might be the error you encountered. 

Best, 

Nicolas

nickilanger, 2019-11-04:
Hi Bene, 

- for me channel 80 is the event channel. I see, the bdi-import function probably by default does not include the event channel. Could automagic throw an error if you try to exclude channels that are not there?

this is good idea. We will implement this. 

- Importing the chanloc with eeglab gave me empty chanlocations.

did you try to BIOSIG toolbox within EEGlab? that's how I did it. 

I also ran the preprocessing with your data, you shared and IC Label ran after some problems with the MEX files:

[image]

The preprocessed file looks actually quiet good. 
[image]

nickilanger, 2019-11-11:
> Could automagic throw an error if you try to exclude channels that are not there?

Done! it's implemented in the development branch.

closed 2020-11-10 by behinger