methlabUZH/automagic #16: [suggestion] two PREP features

behinger, 2019-10-29:
Hi!
I have two suggestions for the PREP pipeline:
- option to disable Clean-Line Noise fitting, currently this is not possible, not even by the PREP options
- save the PREP report. That could potentially be useful (but its quite some time ago that I used PREP, maybe you found it not worth bothering with. Anyway more info => better debugging?)

Cool toolbox btw :-)

nickilanger, 2019-10-29:
Hi Benedikt, 

Similar answer to your previous reported issue with ICLabel. Automagic is primarily a wrapper, which does not intend to modify the original packages and functions. So, you should tell the PREP people to modify this option. Afterwards we are happy to update the new PREP version. :)

I agree with you that it would be nice to have an option to disable the Clean-Line Noise fitting and have a report (we have our own report (see log file), but not specifically for PREP). 

Best, 

Nicolas

behinger, 2019-10-29:
Hi!
That makes sense.
- I found out that the prep report can be generated using "publishPrep" after loading the EEG file. I was just confused because in the "prep"-options the "generate report" was activated but a report never generated.
- Line noise. I understand much better now. I was confused because initially the PREP gui allows setting channels to "0" (but raises errors for other invalid values), but then automagic complains after checking against the prep-default values (saved in prep's "getPrepDefaults")

closed 2020-11-10 by behinger