methlabUZH/automagic #18: Creating issue with EEGLAB graphical interface

arnodelorme, 2019-11-28:
https://github.com/sccn/eeglab/issues/36

We have edited the text on the EEGLAB plugin manager for now, and will eventually remove the plugin from the EEGLAB plugin manager if the problem is not fixed.

JohnSmallridge, 2019-12-02:
Thank you for addressing this and informing us about the problem.
Here's a link to the latest version of Automagic to include as an EEGLAB plugin:
https://github.com/methlabUZH/automagic

nickilanger, 2019-12-02:
Hi Arno, 

We have written multiple emails (e.g. 11/18/2018) to you and to Ramon Cancino that we were trying to submit a new version of Automagic, but there were some problems with the uploading website. Thus the EEGLAB community was using a very old, not fully functioning version of Automagic. However, as John Smallridge stated above, the new version works fine as a EEGLAB plugin. Can you please re-add Automagic as a EEGLAB plugin again? 

Thanks, 

Nicolas

arnodelorme, 2019-12-02:
Hi Nicolas,

Yes, we will figure it out this week. Would you mind to upload the new version?

http://sccn.ucsd.edu/eeglab/plugin_uploader/upload_form.php

We also have an update form which is shorter but it is better if you upload using this form because you might want to update the current description.

What is the problem you were having with this form? We use it on a daily basis.
Best wishes,

Arno

nickilanger, 2019-12-02:
HI Arno, 

Thanks for the fast response. We will upload the new version with the link you provided. 

I will give you feedback and report problems, if any occur. We had those issues with the old uploading webpage before EEGLAB moved to Github.

Best, 

Nicolas

closed 2021-02-17 by ksgfan