methlabUZH/automagic #20: Create issue

Jaseok1004, 2020-04-01:
Hi. there
My name's Jason and I'm from a Korean graduate school student.
I am using the 65 channels and 1000Hz of sampling rate by the EGI system and convert mff to set file. I was set Exclude channels, sampling rate, and channel location (use ced and sfp file by pop_chanedit). then, cleck to create a project button. I have a message that "A list of channel indices separated by space or comma must be given to determine EOG channels".
How about going to the next step? what is the problem?

please. give me a solution.

thank you.

Best,

Jason

JohnSmallridge, 2020-04-06:
Hi Jason,

If you are _not_ specifying the EOG channels for EOG regression then you should make sure the EOG checkbox is not selected in the 'configurations' menu. 
If you _are_ specifying the EOG channels for EOG regression then the channels should be written with channel numbers/indices separated by a space: for example, 1 2 3

I hope that helps, and good luck with your project

Best,
John

Div12345, 2020-11-03:
Hi,
I have a similar problem, it gives me the same error even when EOG regression is not selected in the configurations.

closed 2021-02-17 by ksgfan