methlabUZH/automagic #21: Perform Reference Issue

rosssawka, 2020-04-08:
Hi Automagic Team,

I'm a student with minimal Matlab experience. I had to create my own '.loc' file in EEGLAB for my acquisitions, because I used a 64 electrode Neuroscan quik cap. I keep getting the same error after parpool:
--------------------------------------------------------------------------------------
Warning: [performReference:NoChannelLocations]
Error using performReference (line 69)
evaluation channels must have locations

Error in prepPipeline (line 159)
    [EEG, referenceOut] = performReference(EEG, params);

Error in performPrep (line 97)
[new_EEG, ~, ~] = prepPipeline(new_EEG, prepParams);

Error in preprocess (line 164)
[EEG, EOG] = performPrep(EEG, EOG, PrepParams, EEGSystem.refChan);

Error in Block/preprocess (line 443)
            [EEG, fig1, fig2] = preprocess(data, self.params);

Error in Project/preprocessAll (line 457)
                [EEG, automagic] = block.preprocess();

Error in mainGUI>runpreprocessbutton_Callback (line 753)
    project.preprocessAll();

Error in gui_mainfcn (line 95)
        feval(varargin{:});

Error in mainGUI (line 61)
    gui_mainfcn(gui_State, varargin{:});

Error in
matlab.graphics.internal.figfile.FigFile/read>@(hObject,eventdata)mainGUI('runpreprocessbutton_Callback',hObject,eventdata,guidata(hObject)) 
> In prepPipeline (line 173)
  In performPrep (line 97)
  In preprocess (line 164)
  In Block/preprocess (line 443)
  In Project/preprocessAll (line 457)
  In mainGUI>runpreprocessbutton_Callback (line 753)
  In gui_mainfcn (line 95)
  In mainGUI (line 61)
  In matlab.graphics.internal.figfile.FigFile/read>@(hObject,eventdata)mainGUI('runpreprocessbutton_Callback',hObject,eventdata,guidata(hObject))

--------------------------------------------------------------------------------------------------
I have gotten this error for all three reference channel selections, is this because I'm using a faulty '.loc' file? I had the same issue using a '.ced' file (tried entering 'chanedit' & '.chanedit' file types). I'll zip it and attach it below. 

Beyond this I'm a bit puzzled as to why there is no channel location when performing the channel reference because I have been processing them as .set files with their appropriate channel locations already saved, so I assumed that the '.loc' file, or any channel location file, would be basically redundant.

Thanks!
[Quikcap64.loc.zip](https://github.com/methlabUZH/automagic/files/4451388/Quikcap64.loc.zip)
[Chan locs for Neuroscan 64 quik cap.ced.zip](https://github.com/methlabUZH/automagic/files/4451391/Chan.locs.for.Neuroscan.64.quik.cap.ced.zip)

rosssawka, 2020-04-08:
So I've realized that I was giving the wrong reference electrode (was giving mastoid electrodes rather than the Cz electrode) and that's why referencing was an issue. Additionally, I've read back in previous issues listed in this forum that '.set' files cannot be read, and I believe, due to similar issues with '.cnt' files, that '.cnt' files can't be read too.  I've saved my '.set' files, individually, in my work space as '.mat' files and no longer experience this issue with finding a reference electrode. But, now everything seems to be blocked  after the reference is found and refiltered using PREP. Here is where everything halts:
*******Start preprocessing all dataset*******
Processing file subject 1_Acquisition 12 r ...(file 1 out of 2)
Running Prep...
Using option file in directory ~
Preliminary detrend to compute reference
pop_eegfiltnew() - performing 847 point highpass filtering.
pop_eegfiltnew() - transition band width: 1 Hz
pop_eegfiltnew() - passband edge(s): 1 Hz
pop_eegfiltnew() - cutoff frequency(ies) (-6 dB): 0.5 Hz
pop_eegfiltnew() - filtering the data (zero-phase)
firfilt(): |====================| 100%, ETE 00:00
Line noise removal
Starting parallel pool (parpool) using the 'local' profile ...
Connected to the parallel pool (number of workers: 2).
Find reference
pop_eegfiltnew() - performing 847 point highpass filtering.
pop_eegfiltnew() - transition band width: 1 Hz
pop_eegfiltnew() - passband edge(s): 1 Hz
pop_eegfiltnew() - cutoff frequency(ies) (-6 dB): 0.5 Hz
pop_eegfiltnew() - filtering the data (zero-phase)
firfilt(): |====================| 100%, ETE 00:00

Anyone have an idea regarding what's happening? I'm running Matlab R2020a so I don't believe the issue is caused by the edition of Matlab being used. Additionally, my acquisitions are rather long, around 27 minutes on average (0.3-0.4 Gb size). Really want this amazing program to work!

JohnSmallridge, 2020-04-20:
Hi,
Thanks for being so detailed with your post. 
I would suggest you try using the clean_rawdata method instead of PREP, and let us know if that improves the situation.

rosssawka, 2020-04-20:
No problem! Yeah without PREP it all works just fine. Really wanted to compare finished products between clean_rawdata and PREP, though.

closed 2021-02-17 by ksgfan