methlabUZH/automagic #23: Parameters about deleted data

RiverTOREAD, 2020-05-25:
Dear Automagic team,
        I'm a student interested in EEG. I benifited a lot from Automagic on preprocessing EEG data. However, there is one question that bothers me several days. So I ask you for help.
       PREP is so time-consuming. I did clean_rawdata() instead. My raw EEG data is 120 seconds long. After being preprocessed using Automagic, I found the preprocessed data was about 116 second long.
      What I want to configure out is how could I known what propotions of specific raw EEG data was deleted during preprocessing progress. Can I find it in automagic structure? It's very important for me to align the preprocessed EEG data with my behavioral data using EEG.event through EEGLAB.
     Thanks for your helpful help. I'm looking forward for your prompt reply.
Sincerely,
Haidong

JohnSmallridge, 2020-05-25:
Dear Haidong,
Thanks for your question, and it's great to hear you're benefiting from Automagic. 
I expect that this is happening because of  the Artifact Subspace Reconstruction (ASR) function in clean_rawdata(). I haven't found a way (at this time) to get the ASR function to clearly state which windows of data were removed. We have found that time points rejected by ASR are variable (not easy to reproduce), so I could suggest unticking the "Run ASR with burst criterion" option in the clean_rawdata() configurations to see if this prevents losing any data. 
Hopefully that helps

JohnSmallridge, 2020-05-25:
Dear Haidong,
My colleague found the solution to your problem: 
In the EEG struct, go to EEG.etc.clean_sample_mask 
There you will find a logical vector for time points which have been retained.
Let us know if this resolves your issue

RiverTOREAD, 2020-05-26:
Dear JohnSmallridge,
Thanks very much. Your answer is very correct to my question. I should spend more time to figure it  out all the information in EEG and automagic structures. Thanks again.
Nest wishes, 
Haidong

RiverTOREAD, 2020-05-26:
Sorry to type Best wishes as Nest wishes.

AlexChalard, 2020-06-04:
I didn't use Automagic since few months, but in the last version Automagic when using ICLabel there was no EEG.etc.clean_sample_mask conversly to the MARA option.

These option can be fixed ?

amibahreini, 2020-06-07:
Dear Alex, 
EEG.etc.clean_sample_mask is a result of Artefact Subspace Reconstruction in clean_rawdata() which is disabled by default. So by default, this field will not exist as no time points are removed from the EEG.
I suspect in your case it is not related to your version of Automagic or ICLabel, rather it is missing simply because you have not used this option.
I hope this helps.
Ami

closed 2021-02-17 by ksgfan