methlabUZH/automagic #24: Re-reference after all the preprocessing steps

RiverTOREAD, 2020-06-08:
Hello automagic teams,
Q1: I input the .set files to automagic with selecting No-scalp reference because my EEG data' reference is common. After finishing all the preprocessing steps, in Data Viewer I found an item called Average Reference: off. However, when I clicked the item above to change Average Reference to on state, I found the final EEG data's reference was still common in EEG structure. What I want to configure out is how to do re-reference to CAR (common average reference) in automagic.
Q2: Afer all the prepocessing steps, automagic found some bad chennels to interpolate. However, I manually delete one or several the bad channels which were found by automagic and add other channels which I thought were bad channels. When I click Interpolte All, I found that the final EEG data of manually deleted bad channels found by autoamgic were all the NaNs. How can I recover the channel data with NaN? For standarization, does not automagic allow this operation?
Thanks a lot. I am looking forward for your prompt reply.
Best wishes,
Haidong

RiverTOREAD, 2020-06-08:
Must the above 2 questions  be solved in command line ranther than in GUI?

RiverTOREAD, 2020-06-13:
Thanks Nicolas for your replies. In my Q2, for example , after preprocessing, automagic found four bad channels, such as chan1, chan 4, chan5, chan38. However, I thought chan4 and chan5 were good channels. As a result, I first clicked "turn on" and deselected  the  chan4 and chan5. This operation changed the chan4 and chan5 from blue horizantal lines to white blank horizantal lines.  Then I clicked "turn off", closed Data Viewer and pressed Interpolate. In the final EEG data, chan4 and chan5 were all NaNs. I found that automagic can't allow this operation due to standarization of prerpocessing. What I could do was to modify the parameters before preprocessing so that chan4 and chan5 could be outside  the criterion of bad channels. In this case, after new preprocessing, chan4 and chan5 would not be in bad channels and never need to be interplate any more. However, it was so cumbersome and time-consuming.

nickilanger, 2020-06-17:
Based on what did you decide that these electrodes were actually good electrodes? Yes, you can not manually redefine bad electrodes as good electrodes. The only proper way of doing this is to change the criteria for bad channels at the beginning of preprocessing. The same criteria is then applied to all files. We ty to avoid as much subjectivity in the pre-processing as possible, as this makes results less reproducible.

RiverTOREAD, 2020-06-18:
I agree with you and it's the purpose of automagic. Preprocessing depedent on experience is so subjective.

closed 2021-02-17 by ksgfan