methlabUZH/automagic #32: ICA Weights look strange

behinger, 2020-11-11:
This is what the ICA winv looks like if I plot the EEGset I receive after automagic:
[image]

These are the one I get, when I overwrite the EEG.icaact (winv etc.) with the EEG.etc.beforeICremove.icaact etc.:
[image]

Note that in the latter component 1 & 5 were marked for deletion. But besides that, something went wrong somewhere. The first set of components does not look at all like a usable ICA-decomposition.

I am using the normal automagic cloned from github yesterday

ksgfan, 2020-12-02:
Hi
The problem arose when recomputing 'icawinv' using eeg_checkset after removing bad ICs. 
`icawinv = pinv(EEG.icaweights * EEG.icasphere)`
and
`icawinv = pinv(EEG.icaweights(good_comps,:) * EEG.icasphere)`
are not the same.

We fixed it - now we use 'icawinv' computed directly in pop_subcom -  and updated the master branch. 
Best,
Dawid

behinger, 2020-12-02:
ok cool :-)

closed 2020-12-02 by behinger