methlabUZH/automagic #34: Not a valid prefix

ugpavlov, 2020-11-30:
[image]
[image]
I tried many different paths and folder structures including BIDS and exactly the structure you show in Automagic directory hierarchy, tried to create a project using GUI and command line but the 'Not a valid prefix' error is still there.

ksgfan, 2020-11-30:
Hi
I cannot replicate this error. Which Automagic version are you currently using? Are all the files in dataFolder called sub-001_*?
Best, Dawid

ugpavlov, 2020-11-30:
Hi! I don't know which version it is but I downloaded it yesterday from github. As you can see on the screenshot all files are called sub-001_*.

ksgfan, 2020-11-30:
Do you already have any preprocessed files in the result folder? If yes, how are they called?

ugpavlov, 2020-11-30:
the result folder is empty.

ksgfan, 2020-12-01:
Ok, could you please go to src->block.m, set a breakpoint in the line 1433, run your code again and examine, what value does the prefix get?

ugpavlov, 2020-12-01:
It's empty.
[image]
I've solved the problem by reseting the paths to default. Far from ideal but it worked. Perhaps it was a conflict with EEGLAB but I have to investigate further. Thanks for you help.

closed 2020-12-02 by ksgfan