methlabUZH/automagic #35: Create Project Issue

yxlFromYSU, 2020-12-21:
Hello,
When I create project, it has the tip as follow:
A list of channel indices seperated by space or comma must be given to determine EOG channels
[image]

ksgfan, 2020-12-21:
Hi
apparently you clicked on the 'Exclude Channels' in the main GUI but provided no channels to exclude. 

[image]

Uncheck the 'Exclude Channels' button or provide list of channels to exclude.

Best,
Dawid

yxlFromYSU, 2020-12-21:
Hello
This is the picture about the configure. I did not clicked on the 'Exclude Channels' in the main GUI.

Best
yxl

ksgfan, 2020-12-21:
I edited my comment a few minutes ago, please look it up again. Does it solve the problem?

yxlFromYSU, 2020-12-21:
yes,this problem has been solved.

closed 2020-12-21 by ksgfan