methlabUZH/automagic #36: Error in GUI preprocessAll

julama, 2021-01-12:
I get the following error for `pre-processing`-> `run` in the GUI:

```
Array indices must be positive integers or logical values.

Error in Project/preprocessAll (line 461)
                mexFolder = str(1:mexFolder(end));

Error in mainGUI>runpreprocessbutton_Callback (line 788)
    project.preprocessAll();

Error in gui_mainfcn (line 95)
        feval(varargin{:});

Error in mainGUI (line 61)
    gui_mainfcn(gui_State, varargin{:});

Error in
matlab.graphics.internal.figfile.FigFile/read>@(hObject,eventdata)mainGUI('runpreprocessbutton_Callback',hObject,eventdata,guidata(hObject)) 
Error while evaluating UIControl Callback.
```

What could be the issue?

ksgfan, 2021-01-12:
Hi
that is a problem with IClabel installation. 

Please go to the automagic folder, /src/Project.m and remove the ~ from the Line 453. It should be like this: 
`found = isempty(ICLabelFolderIndex);`

or download automagic from the master branch again - it's already fixed there. 

Best,
Dawid

ksgfan, 2021-01-12:
Hi 
that is a problem with IClabel installation. Are you connected to the internet while running automagic?

Best,
Dawid

julama, 2021-01-12:
Hi Dawid, 
thanks for your help! 
IClabel was not in this folder. After adding it manually I still get the same error (also restarted the application). 
And yes, I was always connected.

ksgfan, 2021-01-13:
Thats strange. Are you using Automagic as a EEGlab extension (https://github.com/methlabUZH/automagic/wiki/Automagic-as-EEGLAB-extension)?  What Matlab version do you have? 
I tested Matlab 2016b, 2018a, 2020a on OS and Linux and everything works fine. 

Could you please go to automagic folder, /src/Project.m, set a breakpoint at line 452 and check, what value does `ICLabelFolderIndex` take? We can also habe a short zoom meeting today and check it together.

avaruuser, 2022-07-29:
Hi, I ran into the same issue. When I set a breakpoint and investigate the `ICLabelFolderIndex` it is empty. Any suggestion on how this can be resolved?

I used the automagic-development version since it includes the ICLabel plugin. I'm using MATLAB version 2019b on MacOS

ksgfan, 2022-08-02:
Hi, I can't reproduce the error on linux, mac or windows. 

Is IClabel in `automagic-master/matlab_scripts/eeglab14_1_2b/plugins/` ? 

Maybe try run `restoredefaultpath` before starting Automagic GUI?