methlabUZH/automagic #37: Confusion over setting up EEGSystem structure

pauldhami, 2021-02-13:
Greetings,

firstly, thank you for the code!

I am having trouble getting everything running from the command line, with constant errors with regards to my handling of EEGSystem. This is what I have done:

1. Using an already cleaned .set file similar to the data I want to clean, I export the channel data as a .ced.
2. I place this .ced file in 'automagic-master/matlab_scripts' (this is where it should go from my understanding from the wiki)
3. These are my parameters for parms.EEGSystem:

```
'EEGSystem',           struct('name', 'Others',...
                                               'sys10_20', 1, ...
                                               'locFile', 'neuroscan64_chanloc', ...
                                               'refChan', struct([]), ...
                                               'fileLocType', '.ced',...
                                               'eogChans', [], ...
                                               'powerLineFreq', []), ...
```

I then get the following error:

```
Error using readlocs>load_file_or_array (line 624)
readlocs(): cannot find the named file or variable, check syntax

Error in readlocs (line 405)
           array = load_file_or_array( filename, g.skiplines);

Error in pop_chanedit (line 834)
                tmplocs = readlocs( chaninfo.filename, 'defaultelp', 'BESA' );

Error in systemDependentParse (line 141)
            [~, EEG] = evalc(['pop_chanedit(EEG, ''lookup'', EEGSystem.sys10_20_file,' ...

Error in preprocess (line 141)
    systemDependentParse(data, EEGSystem, ChannelReductionParams, ...

Error in Block/preprocess (line 517)
            [EEG, fig1, fig2, fig3] = preprocess(data, self.params);

Error in Project/preprocessAll (line 501)
                [EEG, automagic] = block.preprocess();

Error in automagic_test (line 44)
project.preprocessAll();
```

Any suggestions as how to solve this would be greatly appreciated.

Best,
Paul

ksgfan, 2021-02-13:
Hi Paul

Have you tried that:

`
'EEGSystem',           struct('name', 'Others',...
                                               'sys10_20', 1, ...
                                               'locFile', 'neuroscan64_chanloc**.ced**', ...
                                               'refChan', struct([]), ...
                                               'fileLocType', '**ced**',...
                                               'eogChans', [], ...
                                               'powerLineFreq', []), ...
`

?

Best
Dawid

pauldhami, 2021-02-13:
Thank you for your suggestion. That seems to have solved some issues, but I am now receiving the following error as well after using your suggestion:

```
Error using readlocs>load_file_or_array (line 624)
readlocs(): cannot find the named file or variable, check syntax

Error in readlocs (line 405)
           array = load_file_or_array( filename, g.skiplines);

Error in pop_chanedit (line 834)
                tmplocs = readlocs( chaninfo.filename, 'defaultelp', 'BESA' );

Error in systemDependentParse (line 141)
            [~, EEG] = evalc(['pop_chanedit(EEG, ''lookup'', EEGSystem.sys10_20_file,' ...

Error in preprocess (line 141)
    systemDependentParse(data, EEGSystem, ChannelReductionParams, ...
```

Any help would be greatly appreciated!

ksgfan, 2021-02-13:
Could you share data from one subject and the chanlocs file with me? It would be much easier this way.

You can add me to a dropbox folder: Dawid.strzelczyk@uzh.ch

pauldhami, 2021-02-13:
Hi Dawid,

I have shared it via Google Drive (hope that is okay).

Thank you for looking at it!

ksgfan, 2021-02-13:
Sure, that's fine, but my google account is ksgfan13@gmail.com :)

ksgfan, 2021-02-13:
It works on my machine. Are you sure that you set:

`'locFile', 'neuroscan64_chanloc.ced'`

and

`'fileLocType', 'ced',...`

?
best,
Dawid

pauldhami, 2021-02-13:
Hi Dawid,

thanks for checking. I unfortunately am still running into the same error as posted in my second message. I thought it was perhaps related to how I set up my paths, so I even loaded into Matlab the EEGlab data structure (of the file I shared with you) and ran:

`[EEG_out, plots] = preprocess(EEG, params)`

But am still running into the same error.

pauldhami, 2021-02-13:
Hi Dawid,

I did a clean reinstall of Automagic, and everything seems to be working fine now. I must have done something wrong with the paths or something along those lines.

Thank you for your assistance!

ksgfan, 2021-02-13:
Hi Paul,

thats great! If you require any further information, feel free to contact me.

Best,
Dawid

closed 2021-02-13 by ksgfan