methlabUZH/automagic #39: Error with clean rawdata

pauldhami, 2021-02-14:
Greetings,

Sorry for another question, but I am having trouble with the preprocessing of a single continuous resting state EEG file.

I am trying to replicate the different preprocessing pipelines as described in figure 3 of the Automagic Neuroimage paper.

Specifically, I am interested in comparing the use of PREP versus PREP + clean rawdata (with both being followed by EOGr and MARA).

Although just running PREP seems to work, I am getting an error with regards to clean rawdata.

This is my structure and commands: 

`params = struct(...
                 'CRDParams',           struct('ChannelCriterion',     0.85,...
                           'LineNoiseCriterion',   4,...
                           'BurstCriterion',       20,...
                           'WindowCriterion',      0.25, ...
                           'Highpass',             [0.25 0.75]),...
                 'PrepParams',          struct(), ...
                 'HighvarParams',       struct('sd', 25, ...
                                            'cutoff', 100, ...
                                            'rejRatio', 0.5),...
                 'MinvarParams',        struct('sd', 1),...
                 'FilterParams',         struct('notch', struct('freq',  60), ...
                                               'high',  struct('freq',  1, 'order', []),... % Default order for filtering
                                               'low',   struct('freq',  80, 'order', [])),...
                 'InterpolationParams', struct('method', 'spherical'), ...
                 'RPCAParams',          struct([]), ...
                 'MARAParams',         struct('chanlocMap', containers.Map, ...
                            'largeMap',   0, ...
                            'high',       struct('freq', 2.0,...
                                                 'order', []), ...
                                                 'keep_comps', 0),...
                 'EOGRegressionParams', struct(), ...
                 'ChannelReductionParams', struct(), ...
                 'EEGSystem',       struct('name', 'Others',...
                                               'sys10_20', 1, ...
                                               'locFile', 'CANBIND58_chanloc.ced', ...
                                               'refChan', struct([]), ...
                                               'fileLocType', 'ced',...
                                               'eogChans', [], ...
                                               'powerLineFreq', []), ...
                 'Settings',            struct('trackAllSteps', 0, ...
                                        'pathToSteps', '/allSteps.mat',...
                                        'colormap','Default'),...
                 'ICLabelParams',       struct([]),...
                 'TrimOutlierParams', struct()...
                 );
%%

name = 'commandline_project';
dataFolder = '/Users/paul/Desktop/AutomaticEEG/Automagic/';
resultsFolder = '/Users/paulDesktop/AutomaticEEG/Automagic/';
ext = '.set';
Params = params;
VisualisationParams = struct();

project = Project(name, dataFolder, resultsFolder, ext, Params, VisualisationParams, 10000);
project.preprocessAll();
project.interpolateSelected();
`

I get the following error with regards to clean raw_data:

`Error using  \ 
Matrix dimensions must agree.

Error in performEOGRegression (line 34)
eegclean =  eeg - eog * (eog \ eeg);

Error in preprocess (line 279)
EEG = performEOGRegression(EEG, EOG, EOGRegressionParams);

Error in Block/preprocess (line 517)
            [EEG, fig1, fig2, fig3] = preprocess(data, self.params);

Error in Project/preprocessAll (line 501)
                [EEG, automagic] = block.preprocess();`

Any advice as to what I may be doing wrong?

Thank you in advance.
Paul

ksgfan, 2021-02-14:
Hi Paul

that is a good question! We should explain it better in the wiki.

You need to specify the EOG channels for the EOG regression here:

```
'EEGSystem',       struct('name', 'Others',...
                                               'sys10_20', 1, ...
                                               'locFile', 'CANBIND58_chanloc.ced', ...
                                               'refChan', struct([]), ...
                                               'fileLocType', 'ced',...
                                               'eogChans', [], ...
                                               'powerLineFreq', []), ...
```

e.g. 

` 'eogChans', [1 8 125 126], ...`

If EOG is empty, you get the matrix dimension error

```
>> eog = double.empty

eog =

     []

>> eeg = 1:10

eeg =

     1     2     3     4     5     6     7     8     9    10
>> eegclean =  eeg - eog * (eog \ eeg);
Error using  \ 
Matrix dimensions must agree.
```

Let me know, if you have any other issues.

Best,
Dawid

pauldhami, 2021-02-14:
Hi Dawid,

thank you again for your response!

So if my experiment had no EOG recordings, I would then set 'EOGRegressionParams' structure to: struct([])?

I ask because for some reason, I had 'EOGRegressionParams' set to the default (), but there was no error thrown out when using it when just PREP (and not CRD), even though I had 'eogChans' empty.

Thank you again for your help.
Paul

ksgfan, 2021-02-14:
Hi Paul

Your welcome!

Exactly, if 'EOGRegressionParams',  struct(), ...  then the EOG regression is performed. But you need to specify the EOG channels as mentioned above. If 'EOGRegressionParams',  struct([]),...  then EOG regression is skipped and not performed.

The error occurred in performEOGRegression (line 34), not in CRD :) Empty EOG is no problem for PREP and CRD. If EOG list is empty, systemDependentParse.m creates an EOG struct similar to EEG struct, but with no data (line 434).

Best
Dawid

closed 2021-02-17 by ksgfan