methlabUZH/automagic #40: Project creation naming issue

nicoa47, 2021-02-16:
If the project starts with a name that is alphabetically above the entry "Create new project..." (e.g. a project name starting with "B"), the file can no longer be read in by EEGLab . Only after renaming such that the project name comes alphabetically after "Create new project..." can it be read in.

I would test again whether it must also be under "Load an existing project...".

ksgfan, 2021-02-17:
Hi Nicoa

what file can not be read in EEGlab? The .mat file with EEG data? 

I created a project called 'aaa' and it works fine. Could you give me more details?

Best,
Dawid

nicoa47, 2021-02-19:
Hi Dawid,

thank you for you answer.
When we click on "Import Automagic Results" I obtain the following error message:

```
Error using importResultsGUI>load_selected_project (line 194)
Assertion failed.

Error in importResultsGUI>update_and_load (line 135)
handles = load_selected_project(handles);

Error in importResultsGUI>load_state (line 119)
handles = update_and_load(handles);

Error in importResultsGUI>importResultsGUI_OpeningFcn (line 68)
handles = load_state(handles);

Error in gui_mainfcn (line 220)
    feval(gui_State.gui_OpeningFcn, gui_hFigure, [], guidata(gui_hFigure), varargin{:});

Error in importResultsGUI (line 40)
    [varargout{1:nargout}] = gui_mainfcn(gui_State, varargin{:});

Error in pop_import (line 27)
waitfor(importResultsGUI(params));

Error while evaluating Menu Callback.
```

Thank you
Nico

ksgfan, 2021-02-20:
Hi Nico

I found it, that's a bug. It happens with capital letters A, B and C. I fixed it and committed to the development branch. You can copy the code from there (automagic/eeglab_plugin/importResultsGUI.m) and paste into your automagic-master folder. 

Let me know, if you have any other questions.
Best,
Dawid

closed 2021-02-22 by ksgfan