methlabUZH/automagic #44: Warning: Variable 'reduced' was not saved. For variables larger than 2GB use MAT-file version 7.3 or later.

dominikwelke, 2021-04-20:
hi all,

i spotted another small issue related to large files:

```
Warning: Variable 'reduced' was not saved. For variables larger than 2GB use MAT-file version 7.3 or later. 
> In Block/saveFiles (line 1299)
  In Block/preprocess (line 578)
  In Project/preprocessAll (line 501)
  In mainGUI>runpreprocessbutton_Callback (line 809)
  In gui_mainfcn (line 95)
  In mainGUI (line 61)
  In matlab.graphics.internal.figfile.FigFile/read>@(hObject,eventdata)mainGUI('runpreprocessbutton_Callback',hObject,eventdata,guidata(hObject)) 

```

i guess adding the version flag will fix this

best, Dominik

ksgfan, 2021-04-28:
Hi Dominik,

I changed the version flag to 7.3. It seems to be working fine, but please let me know if you find some issues with that.

Best,
Dawid

dominikwelke, 2021-04-28:
hi Dawid,

yes, I had also changed this locally and did not have problems with it so far.
I saw that you found the second occurence of `v6` saving after the channel interpolation - great, this was causing the same issue when reached.

best
Dominik

closed 2021-04-28 by dominikwelke