methlabUZH/automagic #50: Undefined mcolonmex with startup

fry0083, 2021-05-27:
I have downloaded automagic and then I changed my working directory. Then I get the follow error when I run the command _runAutomagic

>> runAutomagic
Undefined function or variable 'mcolonmex'.

Error in mcolon (line 82)
a =  mcolonmex(i1,d,l);

Error in  :  (line 6)
A = mcolon(varargin{:});

Error in fileparts (line 62)
            pathstr = file(1:ind-1);

Error in addAutomagicPaths (line 29)
automagicPath= fileparts(mfilename('fullpath'));

Error in runAutomagic (line 5)
addAutomagicPaths();
 

Please let me know if this is user error or if I am missing something with download and installation process. Thanks

fry0083, 2021-05-27:
I'm currently running matlab 2018b

ksgfan, 2021-06-09:
Hi

hard to say, what could be the problem. Here are some suggestions:
https://ch.mathworks.com/matlabcentral/answers/96005-why-do-i-get-the-error-undefined-function-or-variable

You don't have to install automagic, just download. Let me know if I could fix the problem or have any other question.

Best,
Dawid

closed 2021-11-09 by ksgfan