methlabUZH/automagic #55: Issue with ICA components visualization

ramBrain, 2021-11-09:
Dear developers,

Since the latest update, it is not possible to visualize all the ICA components from the GUI. The data viewer attempts to only display the ones that were not rejected, but there seems to be a mislabeling (hopefully only on the GUI), as also confirmed by ICLabel with the warning message:
```
Warning: The number of ICs do not match the number of IC classifications. This will result in incorrectly plotted labels. Please rerun ICLabel
> In pop_prop_extended (line 175)
```
Also see the attached figure for an example of a clearly mislabelled component.
[image]

I believe that it would be useful to preserve the full information on the ICA step (preserved and rejected components), for the purpose of manual double-checking. At the moment the only information on this step is preserved in the `automagic` variable, where the rejected components are listed, but cannot be visually inspected. Would that be a "quick" fix?

Thank you in advance for your help.

Ramtin

ksgfan, 2021-11-09:
Hi Ramtin,

new version of pop_subcomp removes the rejected components. We fixed it already in the dev version by adding 
`EEG.etc.ic_classification.ICLabel.all_classifications = EEG.etc.ic_classification.ICLabel.classifications;` before calling pop_subcomp in `performIClabel`. 

Next, the `viewICsGUI.m` needs to be properly adjusted: `classifications` to  `all_classifications` (see dev version).

Best,
Dawid

ramBrain, 2021-11-09:
That sounds great, I will just pull the new version. Thanks!

Ramtin

ksgfan, 2021-11-09:
great, but be careful, because we added a new feature (https://github.com/olafdimigen/OPTICAT) and the new Automagic version will not work for old projects. It may be easier to just change these few lines of code.

closed 2021-11-16 by ramBrain