methlabUZH/automagic #58: Error while running a BIDS dataset

koolgax99, 2022-03-24:
Me and my team are working on a dataset for research purposes, while running Automagic we ran into a error. The error is attached as follows. After much debugging we found that theres some issue in the `read_brainvision_vhdr()` function. The function is not able to read the data file. 

System Specifications:
`Windows 10`
`Matlab 2022a`
`Automagic v.2.6`

The dataset which we are using is BIDS compliant and can be found here: [OpenNeuro HD-EEGtask(Dataset 1)](https://openneuro.org/datasets/ds003420/versions/1.0.2)

The screenshot of the error:
[image]

The screenshot of the Automagic settings primarily:
[image]

Please help me with this error.

cc @ksgfan @nickilanger

ksgfan, 2022-03-24:
Hi,

it seems that the fieldtrip function that is part of the old EEGLAB has some problems loading the BrainVision data. 

You could go to the `read_brainvision_vhdr.m` and change line 69
`  info = dir(datafile); `
to
`  info = dir(fullfile(p, '*sub*.eeg'));`

The other option is to use development version of Automagic. We updated there all toolboxes, including EEGLAB. 

Also, from what I see, you will get another error, because you don't specify channel location file. Browse to 
`automagic-master/matlab_scripts/eeglab14_1_2b/plugins/dipfit2.3/standard_BESA/` and select `standard-10-5-cap385.elp`. 
Additionally, specify channel location file type: `.besa`.

The last hint - automagic recognizes BIDS format, so you could change `Data Folder` to `.../research/hd_eeg_tracking`. This way, all subject will be added to the project. 

Best,
Dawid

koolgax99, 2022-03-24:
Hi @ksgfan, 
Thanks for the wonderful answer, but still i am not getting the desired output. 

- `The other option is to use development version of Automagic. We updated there all toolboxes, including EEGLAB.` 
I am going ahead with this option. But there is a new error which Matlab is throwing and it can be seen in the ss attached below.

[image]

- ```Also, from what I see, you will get another error, because you don't specify channel location file. Browse to automagic-master/matlab_scripts/eeglab14_1_2b/plugins/dipfit2.3/standard_BESA/ and select standard-10-5-cap385.elp. Additionally, specify channel location file type: .besa. ```

This suggestion asks me to specify the list of EOG channel indices in a comma or space separated format. Can you please tell me why this error persists?

The screenshot for the same is attached below
[image]

- ```The last hint - automagic recognizes BIDS format, so you could change Data Folder to .../research/hd_eeg_tracking. This way, all subject will be added to the project. ```

Also i tried this too but somehow the Automagic is not detecting my eeg files and overdetecting the number of subjects. I have 23  subjects but it is showing 28 subjects. I have attached the screenshot for your reference. 
[image]

Thankyou for your time.  I really appreciate your help and work around AutoMagic.

ksgfan, 2022-03-24:
File Extension should be still `vhdr`, not `eeg`

The error regarding EOG channels is probably due to EOG regression. If you want to perform EOG regression, you should specify EOG channels in the Configuration tab, just below the `EOG regression` tab(e.g.` 10, 11, 23).

Number of subjects doesn't matter much, because folders without `vhdr` files will be skipped. 

Let me know, if that helps.

koolgax99, 2022-03-25:
Hi @ksgfan, 

I tried all the above configs, the Automagic is working now but it is throwing an error while preprocessing the data.
This is the error i am getting right now in my matlab, after pressing he run button.
```
Processing file eeg_sub-01_ses-naming_run-1_eeg ...(file 1 out of 2)

ans =

    'Matrix dimensions must agree.'

Warning: Escaped character '\s' is not valid. See 'doc sprintf' for supported special characters. 
> In Project/writeToLog (line 1850)
In Project/preprocessAll (line 523)
In mainGUI>runpreprocessbutton_Callback (line 821)
In gui_mainfcn (line 95)
In mainGUI (line 61)
In matlab.graphics.internal.figfile.FigFile/read>@(hObject,eventdata)mainGUI('runpreprocessbutton_Callback',hObject,eventdata,guidata(hObject))
```

[image]

Please help me solve it, thanks!!!

ksgfan, 2022-03-25:
weird, because I tried preprocess your data and it works on my computer. Could you show me the main GUI window again? 
Additionally, could you go to script `automagic/preprocessing/preprocess.m`, set a breakpoint on line 143 and check, whether the `data` was loaded correctly? `data` should be a Matlab `struct` with the EEG.data field.

koolgax99, 2022-03-25:
hi @ksgfan 

This is the main GUI of my Automagic.
[image]

However if its working in your computer can you please share your system specification for running it:
```
  - Matlab Version
  - EEGlab version
  - Automagic (dev or stable)
  - Other dependencies version
```

Thankyou so much for helping!!

ksgfan, 2022-03-25:
Matlab 2020b
EEGlab 2020_0 (it comes with automagic)
Automagic developement 

Also i am using a Mac OS 12. 

But that shouldn't be a problem. Make sure that the data is not empty and that it is loaded properly.

koolgax99, 2022-03-28:
Hi @ksgfan 
I think i got where the issue is, it was because my data was not loaded properly. However, i still have a couple of minute errors. So, can we have a meet (Google Meet or Zoom Call) to help me debug those and also I wanted to know more about Automagic.

Thankyou, looking forward to hearing from you soon.

ksgfan, 2022-03-28:
Yes sure. I am available almost the whole time. Please send me an e-mail with a date that fits you.
dawid.strzelczyk@uzh.ch