methlabUZH/automagic #61: possible enhancement: option to add excluded channels back to data after preprocessing

dominikwelke, 2022-04-30:
hi all,

I am missing an option to exclude misc channels (EOG, ECG, EMG, or the like) only temporarily during preprocessing.
so far this data is removed from the file.

it's easy to fix this later on, but i think it would be an improvement for automagic if the pipeline could handle this.

the simplest implementation would be to store all excluded channel data (in memory or better on disc) and reattach them to the preprocessed file before saving.
this could be controlled with a checkbox in the gui.

alternatives: 
- data could be added back by default (no gui change needed)
- let ppl choose which channels to reattach. not sure if this is necessary, though.

I'd be happy to implement this functionality if you are interested (not the gui, though).
one would need to pay attention to the channel indices. especially if automagic adds a reference channel to the dataset

cheers, 
Dominik

ksgfan, 2022-05-02:
Hi Dominik,

it is a good idea, however as you said, it may be tricky to implement. Especially, when user exclude some channels, add new reference or perform EOG Regression.  For EOG regression, the user has to specify EOG channels that are removed from EEG before preprocessing and regressed out from EEG later.

Additionally, one needs to be careful, when providing channel location file, because if the dimensions of data and channels do not agree, automagic won't load the channel information. 

Up until now, we were simply splitting the data and saving ECG, etc. temporarily on disc (or e.g. in EEG.ecg) before preprocessing in automagic. 

If you are willing to write the code, I am happy to implement the GUI part :) 

Best,
Dawid

dominikwelke, 2022-07-07:
hi @ksgfan 
sorry for the delayed response, i had some busy months in between but finally found the time to work on this :)

> it is a good idea, however as you said, it may be tricky to implement. Especially, when user exclude some channels, add new reference or perform EOG Regression. For EOG regression, the user has to specify EOG channels that are removed from EEG before preprocessing and regressed out from EEG later.

I just created a PR.
lets discuss [there](https://github.com/methlabUZH/automagic/pull/66)

ksgfan, 2022-07-08:
Hi Dominik,

Thank you very much! I am on vacation until 17.07. but will check the PR after i am back. 

Best, 
Dawid