methlabUZH/automagic #64: BIDS-related error after latest submodule update

iustinaabc, 2022-05-20:
Dear developers,

I am currently using Automagic (development-branch) as a submodule in a larger repository. After the latest upgrade of this submodule and its dependencies, there are issues when generating the cleaned-up, preprocessed EEG files.

Specifically, I get this error log:

```
Processing file sub-2022movingtargetpilot04_ses-01_1_256_16_sub-2022movingtargetpilot04_ses-01_1_256_16_epoch07_Cz ...(file 28 out of 35)

ans =

    'Unrecognized field name "isBIDSformat_value".'

Warning: Error in file: C:\Users\u0141695\OneDrive - KU
Leuven\PhD_KUL_backup\PhD_Scientific_work\GitLab_ExpORL\eegorl\utils-and-contrib\automagic\src\Block.m (Line: 1270) ... 
> In Project/preprocessAll (line 507)
In omsi.chain.automagic_toolbox (line 139)
In omsi.main_aadMovingTarget>analyse (line 211)
In omsi.main_aadMovingTarget (line 172) 
Warning: Escaped character '\s' is not valid. See 'doc sprintf' for supported special characters. 
> In Project/writeToLog (line 1852)
In Project/preprocessAll (line 525)
In omsi.chain.automagic_toolbox (line 139)
In omsi.main_aadMovingTarget>analyse (line 211)
In omsi.main_aadMovingTarget (line 172) 

```
The Automagic code used to work before the submodule update, so I wonder what it amiss now?

Thanks a lot in advance!

Kind regards,
Iustina

ksgfan, 2022-05-20:
Hi Iustina,

new automagic versions won't work with old projects. You have to use the old automagic version or create a new project. All versions can be found here: https://github.com/methlabUZH/automagic/commits/development. 

Best,
Dawid

iustinaabc, 2022-05-20:
Hi Dawid,

Are you suggesting to do the preprocessing from scratch?
If so, I already tried this (deleted the project state file and all other outputs of automagic).

Best,
Iustina

ksgfan, 2022-05-20:
Did you also created a new project with new params etc.? 

The error indicates that new automagic version looks for field `isBIDSformat_value`, but this field does not exist in `params`.

iustinaabc, 2022-05-23:
Hi Dawid,

Indeed it seems I was missing the new Automagic parameter  'isBIDSformat_value' in my params file (according to https://github.com/methlabUZH/automagic/blob/development/preprocessing/RecommendedParameters.m).

Unfortunately that was not apparent in the _RecommendedParameters.m_ from the master branch, but after adding it, the preprocessing works!

Thanks for the hint!

Kind regards,
Iustina

closed 2022-05-23 by iustinaabc