methlabUZH/automagic #68: Error when creating a new project

XlZha, 2023-06-25:
Hi,

When I tried to creat a new project, I got the below error:

[image]
 
And the parameters are set as below:
[image]

Could you please tell me how to correct it?

ksgfan, 2023-06-27:
Hi,

what is the name of the file?
could you set a breakpoint on line 1450, before creating the project? Next, create a project and check, what value the `reducedAddress` has?

If you want, I could also assist you via zoom. 

Best,
Dawid

ksgfan, 2023-06-27:
Hi Long,

I think something went wrong, I don't see your screen shots, but only text. 

Best,
Dawid

XlZha, 2023-06-27:
Dear Dawid,

I sent these screen shots as attached.

Best,
Long

closed 2023-07-17 by ksgfan