methlabUZH/automagic #69: Can't reload project

mhershkow, 2023-08-11:
Hi,

I managed to use Automagic once successfuly.

when I tryed to start another project and reload more EEG raw data I get this error:

[image]

Can you help me please?

Moshe

ksgfan, 2023-08-14:
Hi Mosche : )

It seems that Automagic can't find (and open) the `700_test.set` file. Are you sure that the file exists in the `/.../TEST/OLD `folder?

If you still facing this issues, we can organize a zoom meeting and try to solve it together.

Best,
Dawid

mhershkow, 2023-08-17:
Dear Dawid,

Thank you so much for your kind response.

It appears that I didn't use the recommended file setting (wasn't adding a
"subject" folder for each one of my subject's data).

AutoMagic in running excellent right now, I also helped several of my PhD
colleagues with it as well.

I appreciate from the bottom of my heart your precious time and your offer
to meet via Zoom. For now everything seems fine.

I would be happy if I could consult with you in the future about other
questions if any arise.

The holiday season is approaching in Israel and the Hebrew New Year will
soon be upon us. I would like to take this opportunity to wish you and
everyone dear to you a happy and successful new year.

Best wishes,

Moshe

‫בתאריך יום ב׳, 14 באוג׳ 2023 ב-9:50 מאת ‪ksgfan‬‏ <‪
***@***.***‬‏>:‬

> Hi Mosche : )
>
> It seems that Automagic can't find (and open) the 700_test.set file. Are
> you sure that the file exists in the /.../TEST/OLD folder?
>
> If you still facing this issues, we can organize a zoom meeting and try to
> solve it together.
>
> Best,
> Dawid