methlabUZH/automagic #71: Error with channel location (BIDS dataset

eledau, 2024-05-13:
Hello, I have been trying to pre process my data that are stored in BIDS format. These are the settings I am using:
[image]

This is the error I get:
[image]

In each subject folder I have a .tsv file (that is supposed to have the channel information), a .eeg file with raw eeg data, a .json file and .vhdr and .vmrk files. I have also tried to not provide a channel location document or create one myself and save it as .sfp but I keep getting the same error. I also tried to specify one of the subject's .tsv files as channel location file but it didn't work.

ksgfan, 2024-05-13:
Hi, 

it seems that the loading of chanlocs file failed.

if you already have tsv files with electrode locations in each subject folder, you don’t need to provide a new channel location file or file type. Automagic will find these tsv files on its own. Simply remove both entries.

Hope it helps.

eledau, 2024-05-13:
I tried what you suggested but I keep getting the same error, how can I send you the data of 1 subject? Thank you for your help

ksgfan, 2024-05-13:
You can use google drive, Dropbox or wetransfer. I will have a look at the data tomorrow.

eledau, 2024-05-13:
https://drive.google.com/drive/folders/1Hl-91rN7LqsLUMGhqPED-6gRSE6Onuy6?usp=drive_link

ksgfan, 2024-05-14:
thanks!

Could you also provide the '*electrodes.tsv' file or a channel location file that contains locations for 'VPVA', 'VNVB', etc. channels? Brainclinics Foundation

eledau, 2024-05-14:
I do not have that file, but I copied can send you the channel location provided in the paper where the dataset was presented: https://drive.google.com/file/d/1QRsFDnVTCZq3nxiZ68IgnkUm6WU5ESQF/view?usp=drive_link

ksgfan, 2024-05-14:
First, please remove the first line from the .sfp file. This file should look like this:

```Fp1     -26.81  84.06   -10.56
Fp2      29.41  83.74   -10.04
F7      -66.99  41.69   -15.96
F3      -48.05  51.87    39.87
Fz        0.90  57.01    66.36
F4       50.38  51.84    41.33
F8       68.71  41.16   -15.31
FC3     -58.83  21.02    54.82
FCz       0.57  24.63    87.63
FC4      60.29  21.16    55.58
T7      -83.36 -16.52   -12.65
C3      -65.57 -13.25    64.98
Cz        0.23 -11.28    99.81
C4       66.50 -12.80    65.11
T8       84.44 -16.65   -11.79
CP3     -65.51 -48.48    68.57
CPz      -0.42 -48.77    98.37
CP4      65.03 -48.35    68.57
P7_T5   -71.46 -75.17    -3.70
P3      -55.07 -80.11    59.44
Pz       -0.87 -82.23    82.43
P4       53.51 -80.13    59.40
P8_T6    71.10 -75.17    -3.69
O1      -28.98 -114.52    9.67
Oz       -1.41 -117.79   15.84
O2       26.89 -114.68    9.45
VPVA      
VNVB       
HOHL      
HNHR       
Erbs       
OrbOcc     
MASS 
```

Next, use this configurations:
[image]

You have to remove channels that don't have locations on the scalp (i.e., `Exclude Misc. Channels`). Otherwise, PREP and ICA won't work.

I was able to successfully preprocess both files with this settings:
[image]

Let me know if it works for you.

eledau, 2024-05-14:
Thanks! this solved my problem!