methlabUZH/automagic #74: Error Loading .bdf Format Files

zmy1260222041, 2024-12-12:
Dear Developers,

I hope this message finds you well. I am trying to load files in the .bdf format, but I encountered the following error:

matlab.graphics.internal.figfile.FigFile/read>@(hObject,eventdata)mainGUI('runpreprocessbutton_Callback',hObject,eventdata,guidata(hObject)) Skipping preprocessing for this file...

Could you please help me resolve this issue? Thank you for your assistance!

Best regards,
Michael Z

ksgfan, 2024-12-12:
Hi Michael,
Could you provide more details, such as the preprocessing parameters you’re using? Also, could you share the full error message? That will help me better understand the issue.

zmy1260222041, 2024-12-25:
Sure! These are my configurations:
[image]
[image]
The full error message is as follows:
*******Start preprocessing all dataset*******
Processing file my data ...(file 1 out of 30)
Warning: Escaped character '\e' is not valid. See 'doc sprintf' for supported special characters. 
> In Project/preprocessAll (line 510)
In mainGUI>runpreprocessbutton_Callback (line 842)
In gui_mainfcn (line 95)
In mainGUI (line 61)
In matlab.graphics.internal.figfile.FigFile/read>@(hObject,eventdata)mainGUI('runpreprocessbutton_Callback',hObject,eventdata,guidata(hObject)) 
Error in Project/preprocessAll (line 510)
In mainGUI>runpreprocessbutton_Callback (line 842)
In gui_mainfcn (line 95)
In mainGUI (line 61)
In matlab.graphics.internal.figfile.FigFile/read>@(hObject,eventdata)mainGUI('runpreprocessbutton_Callback',hObject,eventdata,guidata(hObject)) 
Error in Project/preprocessAll (line 510)
In mainGUI>runpreprocessbutton_Callback (line 842)
In gui_mainfcn (line 95)
In mainGUI (line 61)
In matlab.graphics.internal.figfile.FigFile/read>@(hObject,eventdata)mainGUI('runpreprocessbutton_Callback',hObject,eventdata,guidata(hObject)) 
Error in Project/preprocessAll (line 510)
In mainGUI>runpreprocessbutton_Callback (line 842)
In gui_mainfcn (line 95)
In mainGUI (line 61)
In matlab.graphics.internal.figfile.FigFile/read>@(hObject,eventdata)mainGUI('runpreprocessbutton_Callback',hObject,eventdata,guidata(hObject)) 
Error in Project/preprocessAll (line 510)
In mainGUI>runpreprocessbutton_Callback (line 842)
In gui_mainfcn (line 95)
In mainGUI (line 61)
In matlab.graphics.internal.figfile.FigFile/read>@(hObject,eventdata)mainGUI('runpreprocessbutton_Callback',hObject,eventdata,guidata(hObject)) 
Error in Project/preprocessAll (line 510)
In mainGUI>runpreprocessbutton_Callback (line 842)
In gui_mainfcn (line 95)
In mainGUI (line 61)
In matlab.graphics.internal.figfile.FigFile/read>@(hObject,eventdata)mainGUI('runpreprocessbutton_Callback',hObject,eventdata,guidata(hObject)) 
Error in Project/preprocessAll (line 510)
In mainGUI>runpreprocessbutton_Callback (line 842)
In gui_mainfcn (line 95)
In mainGUI (line 61)
In matlab.graphics.internal.figfile.FigFile/read>@(hObject,eventdata)mainGUI('runpreprocessbutton_Callback',hObject,eventdata,guidata(hObject)) 
Error in Project/preprocessAll (line 510)
In mainGUI>runpreprocessbutton_Callback (line 842)
In gui_mainfcn (line 95)
In mainGUI (line 61)
In matlab.graphics.internal.figfile.FigFile/read>@(hObject,eventdata)mainGUI('runpreprocessbutton_Callback',hObject,eventdata,guidata(hObject)) 
Error in Project/preprocessAll (line 512)
In mainGUI>runpreprocessbutton_Callback (line 842)
In gui_mainfcn (line 95)
In mainGUI (line 61)
In matlab.graphics.internal.figfile.FigFile/read>@(hObject,eventdata)mainGUI('runpreprocessbutton_Callback',hObject,eventdata,guidata(hObject)) 
Skipping preprocessing for this file... 
Warning: Escaped character '\d' is not valid. See 'doc sprintf' for supported special characters. 
> In Project/writeToLog (line 1859)
In Project/preprocessAll (line 530)
In mainGUI>runpreprocessbutton_Callback (line 842)
In gui_mainfcn (line 95)
In mainGUI (line 61)
In matlab.graphics.internal.figfile.FigFile/read>@(hObject,eventdata)mainGUI('runpreprocessbutton_Callback',hObject,eventdata,guidata(hObject))